oresat/oresat-firmware #24: Radio Engineering Data Link

heliochronix, 2020-08-13:
OreSat Engineering Data Link code for sending and receiving data via the AX5043 drivers.

heliochronix, 2020-10-09:
The "Engineering Data Link" will utilize the HDLC framing engine of the AX5043, which provides frame synchronization, bit stuffing, and a final CRC-16-CCITT check at the end of a frame.

The actual "Transfer Frame" will be implemented according to the "Unified Space Data Link Protocol" [CCSDS 732.1-B-1](https://public.ccsds.org/Pubs/732x1b1.pdf) with some optional "Space Data Link Security protocol" [CCSDS 355.0-B-1](https://public.ccsds.org/Pubs/355x0b1.pdf) security protocols.

The payload for frames will be "Space Packet Protocol" [CCSDS 133.0-B-2](https://public.ccsds.org/Pubs/133x0b2.pdf) packets.

ryanpdx, 2024-06-02:
No longer relevant

closed 2024-06-02 by ryanpdx