oresat/oresat-firmware #51: Can't build app_control for STM32F446RE

sticky-tea, 2021-09-07:
Trying to build app_control for STM32F446RE, doing `make` in oresat-firmware/src/f4/app_control but get this:
`
In file included from ../../../ext/ChibiOS/os/hal/include/hal_efl.h:53:0,
                 from ../../../ext/ChibiOS/os/hal/include/hal.h:144,
                 from ../../../common/include/CO_driver_target.h:34,
                 from ../../../ext/CANopenNode/301/CO_driver.h:32,
                 from ../../../common/CO_driver.c:29:
../../../ext/ChibiOS/os/hal/ports/STM32/STM32F4xx/hal_efl_lld.h:71:2: error: #error "This EFL driver does not support the selected device"
 #error "This EFL driver does not support the selected device"
  ^~~~~
../../../ext/ChibiOS/os/common/startup/ARMCMx/compilers/GCC/mk/rules.mk:189: recipe for target 'build/obj/CO_driver.o' failed
make: *** [build/obj/CO_driver.o] Error 1
`

I found that ChibiOS supports only F412, F413 and F40_41:
`
#if defined(STM32F413xx) || defined(STM32F412xx) || defined(STM32F40_41xxx)  \
                                                 || defined(__DOXYGEN__)
...

#else
#error "This EFL driver does not support the selected device"
#endif

`

So what am I doing wrong and how should I build it?

heliochronix, 2021-09-07:
A temporary patch is needed for ChibiOS until OreSat moves to ChibiOS  21.6.0 (at which point the patch is incorporated upstream). Extract/copy this patch into the `ext/ChibiOS` directory and apply it.
[stm32f42x_43x_efl.zip](https://github.com/oresat/oresat-firmware/files/7122890/stm32f42x_43x_efl.zip)

sticky-tea, 2021-09-07:
> A temporary patch is needed for ChibiOS until OreSat moves to ChibiOS 21.6.0 (at which point the patch is incorporated upstream). Extract/copy this patch into the `ext/ChibiOS` directory and apply [...]

Doesn't work :(
I made the following:
```sh
git clone https://github.com/oresat/oresat-firmware && cd oresat-firmware 
git submodule update --init
cp ~/stm32f42x_43x_efl.patch ext/ChibiOS/os # 'cause ext/ChibiOS got error "error: hal/ports/STM32/STM32F4xx/hal_efl_lld.c: No such file or director"

cd ext/ChibiOS/os
git apply --check stm32f42x_43x_efl.patch

cd ../../../src/f4/app_control
make
```

And got the same error

heliochronix, 2021-09-07:
Sorry, you need to  `patch -p0 < stm32f42x_43x_efl.patch` in `ext/ChibiOS`. This patch was made for a subversion repo (which is what ChibiOS uses upstream, this was the upstream patch submission). When this is done you should see that the git repo is in the following modified state:
```
➜ git status
HEAD detached at d96c2af16
Changes not staged for commit:
  (use "git add <file>..." to update what will be committed)
  (use "git restore <file>..." to discard changes in working directory)
	modified:   os/hal/ports/STM32/STM32F4xx/hal_efl_lld.c
	modified:   os/hal/ports/STM32/STM32F4xx/hal_efl_lld.h

Untracked files:
  (use "git add <file>..." to include in what will be committed)
	stm32f42x_43x_efl.patch

no changes added to commit (use "git add" and/or "git commit -a")
```

sticky-tea, 2021-09-07:
> Sorry, you need to `patch -p0 < stm32f42x_43x_efl.patch` in `ext/ChibiOS`. This patch was made for a subversion repo (which is what ChibiOS uses upstream, this was the upstream patch submission). [...]

Thank you! Now it compiles, but there are some errors with comms.c

```
Compiling comms.c
./source/comms.c:560:5: error: initializer element is not constant
     lband_high_cfg,
     ^~~~~~~~~~~~~~
./source/comms.c:560:5: note: (near initialization for 'radio_cfgs[0]')
./source/comms.c:561:5: error: initializer element is not constant
     lband_low_cfg,
     ^~~~~~~~~~~~~
./source/comms.c:561:5: note: (near initialization for 'radio_cfgs[1]')
./source/comms.c:562:5: error: initializer element is not constant
     uhf_eng_cfg,
     ^~~~~~~~~~~
./source/comms.c:562:5: note: (near initialization for 'radio_cfgs[2]')
./source/comms.c:563:5: error: initializer element is not constant
     uhf_ax25_cfg,
     ^~~~~~~~~~~~
./source/comms.c:563:5: note: (near initialization for 'radio_cfgs[3]')
./source/comms.c:55:24: warning: 'vc1' defined but not used [-Wunused-const-variable=]
 static const uslp_vc_t vc1 = {
                        ^~~
../../../ext/ChibiOS/os/common/startup/ARMCMx/compilers/GCC/mk/rules.mk:189: recipe for target 'build/obj/comms.o' failed
make: *** [build/obj/comms.o] Error 1

```

Ubuntu 18.04, gcc-arm-none-eabi from ubuntu repo

heliochronix, 2021-09-07:
Interesting... I haven't seen that on GCC 10+,

Ubuntu 18.04 ARM GCC has some issues if I'm not mistaken, and will throw warnings/errors when compiling F0 code. Would you be up for trying the current GCC provided by ARM?

https://developer.arm.com/tools-and-software/open-source-software/developer-tools/gnu-toolchain/gnu-rm/downloads

sticky-tea, 2021-09-08:
> Interesting... I haven't seen that on GCC 10+,
> 
> Ubuntu 18.04 ARM GCC has some issues if I'm not mistaken, and will throw warnings/errors when compiling F0 code as well. Would you be up for trying the current GCC provided by ARM?
> 
> https://developer.arm.com/tools-and-software/open-source-software/developer-tools/gnu-toolchain/gnu-rm/downloads

Yeah, it works for arm-none-eabi-gcc 10.3.1. Thank you so much!

closed 2021-09-08 by sticky-tea